MitjaNemec/Archive3DModels #1: 3d models created failed

mwl1994, 2022-03-05:
hello ，my kicad  version is (6.99.0－8g7c91cd73e3)，when I
  use this plugin， I met the following question，the log is

03-05 10:38:12 com_github_MitjaNemec_Archive3DModels.action_archive_3d_models 210:Plugin executed on: 'win32'
03-05 10:38:12 com_github_MitjaNemec_Archive3DModels.action_archive_3d_models 211:Plugin executed with python version: '3.9.6 (default, Sep 27 2021, 21:15:11) [MSC v.1929 64 bit (AMD64)]'
03-05 10:38:12 com_github_MitjaNemec_Archive3DModels.action_archive_3d_models 212:KiCad build version: (6.99.0-834-g7c91cd73e3)
03-05 10:38:12 com_github_MitjaNemec_Archive3DModels.action_archive_3d_models 213:Plugin version: 1.0.3

03-05 10:38:20 com_github_MitjaNemec_Archive3DModels.archive_3d_models 13:Starting to archive 3D models
03-05 10:38:20 com_github_MitjaNemec_Archive3DModels.archive_3d_models 34:Getting 3D models for footprint of: D10
03-05 10:38:20 com_github_MitjaNemec_Archive3DModels.action_archive_3d_models 247:Fatal error when executing Archive 3D Models
Traceback (most recent call last):
  File "C:\Users\Administrator\Documents\KiCad\6.99\3rdparty\plugins\com_github_MitjaNemec_Archive3DModels\action_archive_3d_models.py", line 240, in Run
    not_copied_list = archiver.archive_3d_models(board, remap_missing_models=amm)
  File "C:\Users\Administrator\Documents\KiCad\6.99\3rdparty\plugins\com_github_MitjaNemec_Archive3DModels\archive_3d_models.py", line 39, in archive_3d_models
    nr_models = range(len(models))
TypeError: object of type 'SwigPyObject' has no len()

MitjaNemec, 2022-03-06:
First of all, thanks for reporting this.

As it is, the plugins will not work with 6.99 as python API changes quite a lot. So this is a "bug" that I will not fix. Would you be so kind and tell me how you installed the plugin? Via PCM from remote location, via PCM through "install from file" option or manually? I think that the first two options should not work as the plugin metadata is set so that it should allow installation only on 6.0

MitjaNemec, 2023-02-21:
The plugin should work wieh V7

closed 2023-02-21 by MitjaNemec