MitjaNemec/Archive3DModels #5: .STEP vs .step

topherbuckley, 2023-07-30:
I'm getting a few "did not succeed to copy 3D models" End of log below. 

```
07-30 17:15:17 com_github_MitjaNemec_Archive3DModels.archive_3d_models 147:Did not succeed to copy 3D models!
('J1', '${KIPRJMOD}/ALTIUM_EMBEDDED_MODELS/12401610E4#2A.STEP')
('U1', '${KICAD6_3DMODEL_DIR}/Connector_JST.3dshapes/JST_SH_BM03B-SRSS-TB_1x03-1MP_P1.00mm_Vertical.STEP')
('TH1', '${KICAD6_3DMODEL_DIR}/custom/NXRT15XH103FA1B040.STEP')
```

It looks like this can be resolved by just renaming the .STEP files to .step. Happy to do that for my project, but just thought you might want to add case insensitivity to the file type identifier in the code as a lot of SnapEDA or otherwise online 3d model providers can provide them with .STEP rather than .step (for reasons that elude me, but they do exist regularly from my experience).

MitjaNemec, 2023-07-31:
Firstly thanks for reporting this.

Can you show also first part of the .log or can you at least list the OS version, KiCad version and plugin version you used.

I particularly need OS type as filename paths behave differently regarding case.

topherbuckley, 2023-08-09:
Hello,

Sorry, I don't think I kept a copy of the full log, but I can give you the rest of the info:

OS: Ubuntu 22.04
KiCAD: 7.0.6 ubuntu22.04.01 release build
plugin: 1.0.9

MitjaNemec, 2023-08-11:
This should be fixed in 1.0.10. You can update if you install from file or you can wait until it becomes available in PCM.

Again, thanks for reporting this

commit 32f6c685 (2023-08-11): Handle uppercase extensions

Fixes #5
closed 2023-08-11 by MitjaNemec