MitjaNemec/Archive3DModels #6: Not working

wiserst, 2023-08-21:
The plugin doesn't work for me. In the project directory it makes a subdirectory with 3d model subfiles. But when it starts it reports an error that it can't find the 3d files. With the kicad_pcb file it does nothing.

MitjaNemec, 2023-08-21:
Hi, thanks for the report. Can you please attach the `archive_3d_models.log` file which should be in the project folder, so that I get a bit more info.

wiserst, 2023-08-21:
Hi, thank you for the reply. I have attached the requested file. Thank you,
[archive_3d_models.log](https://github.com/MitjaNemec/Archive3DModels/files/12401971/archive_3d_models.log)

wiserst, 2023-08-27:
Hi, have you tried to find out why it's not working for me?

wiserst, 2023-08-28:
OK thanks. All 3D model paths are valid in the project. All 3D models can be seen on the PCB and the export to STEP is performed for all models.
[image]
[image]

MitjaNemec, 2023-08-29:
Hmmm, can you open the .kicad_pcb file with text editor and search for `fp_text reference "SMA2"`. Then look for the line where the 3D model is encoded as in:

[image]

wiserst, 2023-08-29:
Yes, here is the whole footprint block "SMA2"

  (footprint "Connectors:J_THT_SMA_EDGE-S" locked (layer "F.Cu")
    (tstamp fe0c18fa-53e6-4080-bed9-204990ec35ea)
    (at 100.3 120.4)
    (property "LCSC" "")
    (property "Manufacturer" "RF Solutions")
    (property "Manufacturer Part Number" "CON-SMA-EDGE-S")
    (property "Sheetfile" "Sheet_RF_Frontend.kicad_sch")
    (property "Sheetname" "RF_Frontend")
    (property "ki_description" "THT SMA 50 Ohm PCB Edge")
    (property "ki_keywords" "Connector")
    (path "/47f20fd4-7332-4804-ac1f-a290cad346c9/44ee0b9d-8f0e-4749-bb89-8dcfaeec61c5")
    (attr through_hole)
    (fp_text reference "SMA2" (at 1.6 -4.14 unlocked) (layer "F.SilkS")
        (effects (font (size 0.6 0.6) (thickness 0.15)))
      (tstamp 2cf866be-6acb-423a-b750-ede83588ceb3)
    )
    (fp_text value "CON-SMA-EDGE-S" (at 2.2 -3.5 unlocked) (layer "F.Fab")
        (effects (font (size 0.25 0.25) (thickness 0.05)))
      (tstamp 2a8bd36b-e184-4c99-9292-cc6f8d2d5a25)
    )
    (fp_text user "${REFERENCE}" (at 2.2 -3.9 unlocked) (layer "F.Fab")
        (effects (font (size 0.25 0.25) (thickness 0.05)))
      (tstamp 454b1a1e-d636-4fd6-9579-a22e89cbc32c)
    )
    (fp_line (start 0.2 -3.6) (end 5.2 -3.6)
      (stroke (width 0.15) (type default)) (layer "F.SilkS") (tstamp dbfc525c-3b36-413b-aa6f-1ad1508b7bd5))
    (fp_line (start 5.2 -3.6) (end 5.2 3.6)
      (stroke (width 0.15) (type default)) (layer "F.SilkS") (tstamp 56f59099-1d11-447f-9ef2-3f819f628056))
    (fp_line (start 5.2 3.6) (end 0.2 3.6)
      (stroke (width 0.15) (type default)) (layer "F.SilkS") (tstamp a6ff51e5-41a1-4e29-a6a1-282f2151e9f6))
    (fp_line (start -9.5 -2.75) (end -8.5 -2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 62bd2fdd-e4e5-4e43-b468-3287b1d3bbdf))
    (fp_line (start -9.5 2.75) (end -9.5 -2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp fea2b645-d8f2-492d-baec-179316435f27))
    (fp_line (start -8.5 -2.75) (end -8.15 -3.1)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp c14eed52-b8ce-430a-ac9d-800be67890a5))
    (fp_line (start -8.5 2.75) (end -9.5 2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 4ec77795-e78d-4a62-a7ba-1df88a0cf339))
    (fp_line (start -8.15 -3.1) (end -4.9 -3.1)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp ad4d3fb6-db12-4f8b-a873-e403b04578d9))
    (fp_line (start -8.15 3.1) (end -8.5 2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp e7245de3-5406-4165-95b3-1738b0b9f0c9))
    (fp_line (start -4.9 -3.1) (end -4.55 -2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 12bbff65-1a63-44fc-8015-43b3a62dbeed))
    (fp_line (start -4.9 3.1) (end -8.15 3.1)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp d64a300e-635c-4bdf-8953-055df308cb1c))
    (fp_line (start -4.55 -2.75) (end -1.65 -2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 562398bb-f58d-4e39-9468-ab3de4571785))
    (fp_line (start -4.55 2.75) (end -4.9 3.1)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp e0fcdb82-95ee-4ed5-8a40-ae66b0654626))
    (fp_line (start -1.65 -3.25) (end 0 -3.25)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp b5c6ed94-0b11-49c6-86e5-df0e23818d03))
    (fp_line (start -1.65 -2.75) (end -1.65 -3.25)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp c6385647-f621-432b-8be1-894295beb659))
    (fp_line (start -1.65 2.75) (end -4.55 2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 17ff8177-052a-45d5-9ac0-27cb8874e670))
    (fp_line (start -1.65 3.25) (end -1.65 2.75)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 08f1b55c-a140-48b8-a6f2-1a365623b489))
    (fp_line (start 0 -3.25) (end 0 -3.6)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp d52d9b9f-25f3-4413-8ab8-e5f236fa4469))
    (fp_line (start 0 3.25) (end -1.65 3.25)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 22a932d3-3e06-43ae-a3b8-dea5f6bfaac0))
    (fp_line (start 0 3.6) (end 0 3.25)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp 18803c71-6917-4a8d-90f6-894da7246165))
    (fp_line (start 0 3.6) (end 5.2 3.6)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp ae65590a-b69f-42e9-8123-5418dd1a253a))
    (fp_line (start 5.2 -3.6) (end 0 -3.6)
      (stroke (width 0.05) (type default)) (layer "F.CrtYd") (tstamp f5e09b5d-2424-4d21-9d7e-937962c39adf))
[... 47 lines cut ...]
      (stroke (width 0.1) (type default)) (layer "F.Fab") (tstamp 8946b8e4-2f53-4983-ab77-d4dd4ce938a0))
    (fp_line (start 4 2.35) (end 4 3.25)
      (stroke (width 0.1) (type default)) (layer "F.Fab") (tstamp 59896056-c3a5-41b5-95df-e678f029da0d))
    (fp_line (start 4 3.25) (end 0 3.25)
      (stroke (width 0.1) (type default)) (layer "F.Fab") (tstamp d5dc7d48-138f-4c0f-bd35-39bbc9a83708))
    (fp_arc (start 3.55 -0.45) (mid 4 0) (end 3.55 0.45)
      (stroke (width 0.1) (type default)) (layer "F.Fab") (tstamp 86312005-d57e-4ecb-87c0-e9c2641cc33d))
    (pad "1" smd roundrect locked (at 2.7 0) (size 4.5 1.8) (layers "F.Cu" "F.Mask") (roundrect_rratio 0.25)
      (net 215 "Net-(R79-Pad1)") (pintype "passive") (tstamp 9fe13cdf-5eb0-420f-9e2d-a1c4b7adf403))
    (pad "1" smd roundrect locked (at 2.7 0) (size 4.5 1.8) (layers "B.Cu" "B.Mask") (roundrect_rratio 0.25)
[... cut ...]
    (pad "2" smd roundrect locked (at 2.75 2.5) (size 4.5 1.8) (layers "F.Cu" "F.Mask") (roundrect_rratio 0.25)
      (net 2 "GND") (pintype "passive") (tstamp c642dc77-4191-41ca-adeb-e3e0d141d248))
    (pad "2" smd roundrect locked (at 2.75 2.5) (size 4.5 1.8) (layers "B.Cu" "B.Mask") (roundrect_rratio 0.25)
      (net 2 "GND") (pintype "passive") (tstamp a83b7d87-fee0-46d3-9572-5391ecf3f6bf))
    (model ":KICAD7_3DMODEL_DIR:Connectors/J_THT_SMA_EDGE-S.STEP"
      (offset (xyz -0.06 0 -0.85))
      (scale (xyz 1 1 1))
      (rotate (xyz -90 0 90))
    )
  )

MitjaNemec, 2023-08-29:
Ok, this clears things up a bit. The path is encoded differently to what the plugin expects. 

Where did you get this footprint? Did you create it by yourself? Did you enter the 3D model path by yourself?

Before I start fixing this, I'll have to find out if this kind of encoding is "legal" at all. If this is not "legal" then I'll probably not even try to fix this as my experience show that fixes to somebody else's errors tend to bite back.

So for you, the fastest solution is to correct/modify the 3D model path for the offending footprints.

And thanks for reporting all of this

wiserst, 2023-08-29:
I use my own libraries (symbols, footptints, 3D models) that I create and maintain myself. I have them stored in a different place than the libraries supplied with KiCad. I configured all settings, including paths, through the GUI in KiCad in a legal way. After all, KiCad works with my libraries including 3D models completely normally. So I don't understand. I am using the latest current stable version of KiCad (7.0.7). Thank you for solving the problem. I would like to use your plugin to distribute my projects.

wiserst, 2023-08-29:
I also see that in your case the path start is read from the global path setting to 3d models from ${KISYS3DMOD} and in my case it is from :KICAD7_3DMODEL_DIR:

That's interesting that KiCad uses different encodings. It might be worthwhile to have your plugin recognize this and be able to work with both variants, or first figure out why this is the case. Let me know why that is. Thank you.

wiserst, 2023-08-29:
I looked again in my kicad_pcb file and there are two different path definitions to the 3D model:

model "${KICAD7_3DMODEL_DIR}/..."
model ":KICAD7_3DMODEL_DIR:..."
	
So I really don't know why. I don't like that.

MitjaNemec, 2023-08-29:
Yeah, the issue is just how the variable is encoded. By my knowledge variables can be encoded either st `$(ENV_VAR)` or `${ENV_VAR}`. This is the first time I've seen the `:ENV_VAR:`  encoding and I don't know how this came to be. Did you enter the whole path manually? It would really be nice to get to the bottom of this.

As colon `:` is forbidden character for file paths on windows so I suspect that KiCad strips illegal characters first, then it resolves for the environment variables, but I don't know how would this behave on Linux or Mac.

wiserst, 2023-08-29:
I tried in the current version of KiCad to open the footprint "SMA2" (and others) in the editor and re-save the selected 3d model and the path was written (overwritten) with start ${KICAD7_3DMODEL_DIR}

So I don't know where the photprints with path :KICAD7_3DMODEL_DIR came from:
I tried a test version earlier, so maybe that's what happened.

It's interesting that KiCad doesn't mind. In any case, I'll fix the paths for the footprints. Don't fix your plugin. 

Sorry, thank you for your time!

closed 2023-08-29 by wiserst

MitjaNemec, 2023-08-29:
I also don't have a clue how this came to be, as Kicad auto replaces environment variables if you browse to the 3D model location.

It could have been a bug in the KiCad code which has since been fixed. This has happened before, a bug got introduced in 6.0.x and fixed in 6.0.x+2.

Anyhow, thanks again for persistence

wiserst, 2023-09-04:
The bug was in the test version of kicad-nightly. I found it by releasing a fix commit.
https://gitlab.com/kicad/code/kicad/-/commit/e7d6c84aeffc9f1f40a319a2528e5d8e88ab8e05