MitjaNemec/Archive3DModels #8: The plugin archive only one 3d model when there is more than one 3D models

mymake-codes, 2024-05-10:
In archive_3d_model.py at line 56 changing:

model = models.pop()

to 

model = models[index]

The plugin works correctly.

Great plugin!! 
Thanks,
Giovanni

MitjaNemec, 2024-05-11:
Thanks for reporting this. It seems that Kicad internals changed a bit.

Thanks also for the solution, although IIRC I had to use 'pop' and 'push' because from python I could not modify models when they were on the list.

Your solution might seem to work bi I suspect that when running the plugin specific footprint will hev more modeles defined than bere running the plugin. So I'll have to test this thoroughly.

MitjaNemec, 2024-05-13:
This should be fixed in [1.0.12](https://github.com/MitjaNemec/Archive3DModels/releases/tag/1.0.12)

MitjaNemec, 2024-07-29:
@mgroz could you be more specific. This issue is about one footprint having multiple 3D models and should be resolved. 
Can you attach the archive3edmodels.log file

MitjaNemec, 2024-07-30:
Yeah KiCad as a pure software package and KiCad official libraries are not really two separate entities. Due to the way data is encoded and due to the fact the KiCad packages libraries by default, both of them are connected. So you have some code in KiCad that assumes kicad libraries. And what you experience is part of these issues.

But I can add an extra warning when detecting non defined variable `KICAD?_3DMODEL_DIR`.

Can you be a bit more verbose about:
> In addition, it did not map my main {MY_LIBS} variable. This pointed to other footprints and 3d models.

Was/is there an issue with the plugin?